bigtreetech/BIGTREETECH-SKR-V1.1 #10: 2004A display

eljuligallego, 2020-06-01:
Hi,
I would like to know is it's possible to use this display with this board and how should be the configuration file to make it works.

from now with the default configuration I'm only able to see black rectangles with the backlight on.

Thanks.

Dakkaron, 2021-01-05:
It is possible but you are at the wrong location.

You will need to configure and recompile Marlin (if that's the firmware you want to use) for that to work.

https://marlinfw.org/

If you are having trouble, I would recommend the Marlin Discord Channel:
https://github.com/MarlinFirmware/Marlin/issues/11846

This repo here is dead. bigtreetech only uses GitHub as a kind of free filehoster. They don't really do issues, commits or pull requests.

eljuligallego, 2021-01-05:
thanks for reply since none in the company seems to answer...

I want to use the smothie firmware since is the firmware configured by default and the only option I found is to use the universal adapter with arduino board but finally bought another display so does matters. 

never mind, thanks again and happy new year.

Dakkaron, 2021-01-05:
I can't help you with the smoothie firmware. I've only used Marlin, which has great support over there. Don't expect any kind of aftermarket support from Bigtreetech.