bigtreetech/BIGTREETECH-SKR-V1.1 #11: Error compiling marlin 2.0 skr v1.1 version in platformio

sqweezr, 2020-10-10:
the marlin 2.0 firmware does not compile with the current version of platformio. Does anyone have a new version which works? i get an error message, which i do not get with the current official release of marlin. As i am a noob and not capable of configuring a stock marlin to work with the SKR v1.1 board i am hoping someone here can help me out........
thanks in advance!

Koen

canihavesomecoffee, 2020-12-19:
What kind of error are you getting?

StefanoTesla, 2024-01-20:
I had problem Invalid value: '1' is not one of 'off', 'chain', 'deep', 'chain+', 'deep+'. for `lib_ldf_mode` option in the `env:STM32F1` section (Control how Library Dependency Finder (LDF) should analyze dependencies (`#include` directives))

I change lib_ldf_mode  = 1 to lib_ldf_mode  = chain in the platformio.ini